efahl/owut #6: Validate versions after downloading completed build

efahl, 2024-06-07:
https://github.com/openwrt/packages/pull/24324#issuecomment-2148595500 item 4

> You should validate the core version as well as package versions returned by the server before proceeding to install. The user has confirmed a certain action, and you need to make sure that this is the action you are going to perform. It is unfortunately not so uncommon that the versions requested and versions delivered don't match, in that case you should abort with an error.

efahl, 2024-06-13:
@dangowrt, what specifically are you thinking on this one?

1) Compare requested release (e.g., `r26608-b2a84c86e3`) to the one in the build manifest.  If different, fail verification.  Pretty sure imagebuilder is already doing this one, if you specify an older version then you get `Error: Received incorrect version r26474-87d5d38e2a (requested r26467-50e26e6fd7)` message back from the ASU/image builder.

2) For each package in the build request (`"dnsmasq-full": "2.90-r1"`), verify that what we have in `owut`'s package list is the same as what came back in the manifest.

Others?

dangowrt, 2024-06-13:
Checking for packages (explicitly) requested but missing in the resulting manifest would be another thing which comes to mind (I don't think I implemented that in `auc`, but it would be nice to have).

efahl, 2024-06-13:
I've added those checks, hacked in some test cases by changing the response and now report:
```
ERROR: Firmware revision mismatch: expected r26608-b2a84c86e3, but got r26608-b2a84c86e3xxx
ERROR: Firmware package version mismatch: 'dnsmasq-full', expected 2.90-r2, but got 2.90-r1
ERROR: Firmware missing requested package: 'htop'
```
I might fiddle with the exact wording before I commit...

efahl, 2024-06-13:
I've done maybe a dozen builds on two different routers using commit https://github.com/efahl/owut/commit/04d081f12a48e97cd8dd01a8b71e682d4a01c636, with above changes and a few more.  Some tests had induced errors, some were without, all worked as expected.

closed 2024-06-13 by efahl