efahl/owut #10: Default package analysis should show "what provides" info

efahl, 2024-06-13:
As of https://github.com/efahl/owut/commit/0c7b579de421f43c591cf36d661756026c71b6e0 the default package analysis changed from:
```
Default package analysis:
  Default              Provided-by
  dnsmasq              default package replaced/provided by dnsmasq-full
  kmod-dwmac-intel     Warning: default package is not present
  libustream-mbedtls   Warning: default package is not present
  nftables             default package replaced/provided by nftables-json
```
to showing
```
Default package analysis:
  Default              Provided-by
  dnsmasq              not installed, possibly replaced by another package
  kmod-dwmac-intel     not installed, possibly replaced by another package
  libustream-mbedtls   not installed, possibly replaced by another package
  nftables             not installed, possibly replaced by another package

```
In the previous version, the package database was consulted to see if the default package is being provided by another installed package, allowing for a much better understanding of the installation.

efahl, 2024-06-28:
Functionality restored in https://github.com/efahl/owut/commit/1ed0c8d6c61988b400d963ba2a5efb857d3efdf6 by running `opkg` and `apk` commands.

Still needs a proper solution as proposed above.

dioni21, 2025-12-30:
Is this a bug in the current solution?

`WARNING: There are 1 missing default packages, confirm this is expected before proceeding`

And:

```bash
# owut list | xargs -n 1 echo | sort | grep -E 'nftables|wpad'
-nftables
-wpad-basic-mbedtls
wpad-mesh-mbedtls
# opkg whatprovides wpa-supplicant
What provides wpa-supplicant
    wpad-mesh-mbedtls
# opkg whatprovides hostapd
What provides hostapd
    wpad-mesh-mbedtls
# opkg whatprovides nftables
What provides nftables
    nftables-json
# 
```

Note that `nftables-json` is NOT listed in `owut list`