efahl/owut #11: Handle missing rpcd-mod-rpcsys

dangowrt, 2024-06-13:
Current situation:
```
Reference error: left-hand side expression is null
In collect_packages(), file /usr/bin/owut, line 721, byte 38:
  called from function collect_all (/usr/bin/owut:1090:19)
  called from anonymous function (/usr/bin/owut:1616:15)

 `    packages.installed = sort(installed.packages);`
  Near here ------------------------------^
```

Expected output:
```
rpcd-mod-rpcsys is missing. Please install it.
```
(or something like that)

efahl, 2024-06-13:
Hmm, seems like it should already be there as a (transitive) requirement of `attendedsysupgrade-common`, which is in the Makefile's `DEPENDS` list...

Should I be doing run-time checking, too?

```
$ opkg whatdepends rpcd-mod-rpcsys
Root set:
  rpcd-mod-rpcsys
What depends on root set
        attendedsysupgrade-common 8     depends on rpcd-mod-rpcsys
```

efahl, 2024-06-18:
I added thorough checks on all `ubus.call`s, generating diagnostics that would have pinpointed both this one and #12 right away.

Fixed in b0b0117a1556e3b73fdd7b26006a152615d5fb8a

closed 2024-06-18 by efahl