efahl/owut #12: SHA256 validation fails for some devices

dangowrt, 2024-06-17:
```
root@unifi6lr:/tmp# owut upgrade
Board-name     ubnt_unifi-6-lr-v1-ubootmod
Target         mediatek/mt7622
Root-FS-type   squashfs
Sys-type       sysupgrade
Package-arch   aarch64_cortex-a53
Version-from   SNAPSHOT r26450-6510eb3b5d (kernel 6.6.30)
Version-to     SNAPSHOT r26657-531b3f667c (kernel 6.6.33)
Build-FS-type  squashfs
Build-at       2024-06-16T22:10:53Z
Image-prefix   openwrt-mediatek-mt7622-ubnt_unifi-6-lr-v1-ubootmod
Image-file     openwrt-mediatek-mt7622-ubnt_unifi-6-lr-v1-ubootmod-squashfs-sysupgrade.itb
Image-URL      https://mirror-03.infra.openwrt.org/snapshots/targets/mediatek/mt7622/openwrt-mediatek-mt7622-ubnt_unifi-6-lr-v1-ubootmod-squashfs-sysupgrade.itb
Installed      127 packages
Top-level       71 packages
Default         36 packages
User-installed  41 packages (top-level only)

[...]
Failures don't affect this device, details at
  https://mirror-03.infra.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/

Requesting build ----------------------
Hash:   8cbde7e0fe38913337516d48bd115e8e
Status: 200
Detail: done

Build completed in 1 seconds.
  rootfs_size_mb = default
  init-script    = no-init-script

Image source: https://sysupgrade.openwrt.org/store/8cbde7e0fe38913337516d48bd115e8e/openwrt-623762341629-mediatek-mt7622-ubnt_unifi-6-lr-v1-ubootmod-squashfs-sysupgrade.itb
Image saved : /tmp/firmware.bin
Manifest    : /tmp/firmware-manifest.json
Verifying   : /tmp/firmware.bin (8913753 bytes) against /tmp/firmware.sha256sums
ERROR: sha256 doesn't match:
    calculated 'null'
    saved      '32c91a29479f145bbf10bf6c17b1b73ec9fd048a4ed5382cc104a7fcb539566e'
```

efahl, 2024-06-17:
Hmm, I'm deleting the downloaded firmware if any of the verification steps fail, so maybe I'm deleting it too soon?  Checking now...

dangowrt, 2024-06-17:
It also fails with the same error when `--keep` is set. What I meant is that the file seems to be downloaded correctly and only sha256 verification for some reason fails, and also only on some devices...

efahl, 2024-06-17:
Got it, thanks.  

The culprit must be busybox sha256sum somehow erroring out (and I'm throwing away the ubus stderr info right now).  I've got an mt7622, so I'll try it there, see if I can reproduce your ubnt's behavior.

efahl, 2024-06-17:
Bah, works fine on mine.  If you've got time, could you add a print line to `/usr/bin/owut`, probably around line 205, run `owut download` and show me the output?
```
        sum: function(file) {
                // Return the checksum for the specified file.
                let data = _run("/bin/busybox", [ "sha256sum", file ]);
printf(">>>>>>>>>>>>>>>>>>>>\n%s\n%.4J\n<<<<<<<<<<<<<<<\n", file, data);
                return data?.code == 0 ? substr(data.stdout, 0, 64) : null;
        },
```

efahl, 2024-06-17:
Don't bother with above, I've just added legitimate diagnostics in a70f7e5cda15eb0b0a6574eea25c7fe82410fd57 that should tell us what's going on with that call...

dangowrt, 2024-06-17:
Trivial reason: I was missing `rpcd-mod-file`, which isn't a dependency nor indirect dependency of `owut` at the moment, it should be added.

efahl, 2024-06-18:
Aha, thanks!  I don't know how I missed that.   I just checked and rpcd-mod-file got sucked in by luci and friends on all my test machines, so I missed that.

Just pushed it to https://github.com/openwrt/packages/pull/24324

closed 2024-06-18 by efahl