efahl/owut #19: List command gets truncated package name.

pcartwright81, 2024-08-17:
miniupnpd-nftables seems to be listed incorrectly.  It is a dep for luci-app-upnp, so I'm not sure why it's there.  Maybe installed it manually.

```
owut list | sed 's/ /\n/g' | grep nftables
-nftables
```

```
opkg list_installed | grep nftables
miniupnpd-nftables - 2.3.7-r1
nftables-json - 1.0.9-r1
```

efahl, 2024-08-17:
That's a documentation shortcoming.  What you are seeing actually means "remove nftables", the prefixing dash is ImageBuilder's syntax for package removal.

I added a chunk of description to https://openwrt.org/docs/guide-user/installation/sysupgrade.owut#list_formatting, what do you think?  Does it need better examples?

pcartwright81, 2024-08-17:
Documentation and examples look great.  Thank you for the clarification!

closed 2024-08-17 by pcartwright81