efahl/owut #20: Now owut lags behind sysupgrade server and mirror server version

dangowrt, 2024-08-28:
```
root@pfusch:~# owut check
Target         mediatek/mt7623
Profile        bananapi_bpi-r2
Root-FS-type   squashfs
Sys-type       sysupgrade
Package-arch   arm_cortex-a7_neon-vfpv4
Version-from   SNAPSHOT r27055-e6fec638d2 (kernel 6.6.43)
Version-to     SNAPSHOT r27230-a70555fb4f (kernel 6.6.47)
Build-FS-type  squashfs
Build-at       2024-08-26T18:30:14Z
Image-prefix   openwrt-mediatek-mt7623-bananapi_bpi-r2
Image-URL      https://downloads.openwrt.org/snapshots/targets/mediatek/mt7623
Image-file     openwrt-mediatek-mt7623-bananapi_bpi-r2-squashfs-sysupgrade.itb
Installed      333 packages
Top-level      122 packages
Default         37 packages
User-installed  99 packages (top-level only)
```

However, in https://downloads.openwrt.org/snapshots/targets/mediatek/mt7623/version.buildinfo
```
r27231-b870c16534
```

This results in the wrong version being requested and subsequent error:
```
Status: 500
Detail: Error: Received incorrect version r27231-b870c16534 (requested r27230-a70555fb4f)
```

dangowrt, 2024-08-28:
@aparcar

efahl, 2024-08-28:
@dangowrt What version of `owut` are you running?  I switched over to use the DL server as the source for version code some weeks ago, specifically because of this issue.   https://github.com/efahl/owut/commit/e329cb9b8303ac102f1d398eb592a86c52e09a73#diff-5d36e8f4e0a8ae4150e600965392f38819b69e78734e9d73c9d7cad5aed6c0e6R1213  

Got released in owut 2024.07.31:

```
$ owut check --device mediatek/mt7623:bananapi_bpi-r2
Target         mediatek/mt7623
Profile        bananapi_bpi-r2
Root-FS-type   squashfs
Sys-type       sysupgrade
Package-arch   arm_cortex-a7_neon-vfpv4
Version-from   SNAPSHOT r26741-dcc4307205 (kernel 6.6.34)
Version-to     SNAPSHOT r27231-b870c16534 (kernel 6.6.47)
```

efahl, 2024-08-28:
I've got a little shell script that grabs the three sources of version code and the ASU server is consistently one build behind.  More over here https://github.com/openwrt/asu/issues/853#issuecomment-2269218111

```
$ ./version_code.sh snapshots mediatek/mt7623 bananapi_bpi-r2
2024-08-28T18:48 mediatek/mt7623 bananapi_bpi-r2 snapshots
  r27231  Mirror3    https://mirror-03.infra.openwrt.org/snapshots/targets/mediatek/mt7623/profiles.json
  r27231  DL  Server https://downloads.openwrt.org/snapshots/targets/mediatek/mt7623/profiles.json
  r27230  ASU Server https://sysupgrade.openwrt.org/json/v1/snapshots/targets/mediatek/mt7623/bananapi_bpi-r2.json
```

dangowrt, 2024-08-28:
```
root@OpenWrt:~# opkg upgrade owut
Upgrading owut on root from 2024.07.24~d8af324c-r1 to 2024.08.09~8a6becd8-r1...
Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/packages/owut_2024.08.09~8a6becd8-r1_all.ipk
Configuring owut.
```

Now new problems. Since `asu` moved things into Docker containers. Using namespaces, cgroups, seccomp, nonewprivs, ... is a good idea when it comes to process isolation, but apparently doing that in the way it is done with Docker breaks things down the road. Now:
```
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/targets/mediatek/mt7623/kmods/6.6.47-1-1803cefc2ce94661be213c67a2405e89/kmod-usb-storage_6.6.47-r1_arm_cortex-a7_neon-vfpv4.ipk, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_install_pkg: Failed to download kmod-usb-storage. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package kmod-usb-storage.
 * opkg_install_pkg: Checksum or size mismatch for package kmod-usb-storage. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package kmod-usb-storage-extras.
 * opkg_install_pkg: Checksum or size mismatch for package kmod-usb-storage. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package kmod-usb-storage-uas.
make[2]: *** [Makefile:220: package_install] Error 255
make[1]: *** [Makefile:161: _call_manifest] Error 2
make: *** [Makefile:322: manifest] Error 2

ERROR: Build failed with status 500
```

But that's another problem...
```

closed 2024-08-28 by dangowrt