efahl/owut #21: ERROR: Sysupgrade type 'combined-efi' should be one of [ "combined", "rootfs" ]

AlbrechtL, 2024-11-09:
I don't know if this is really a bug but `owut` told me to report it ;-) .

```
root@OpenWrt:~# owut check --version-to SNAPSHOT
owut - OpenWrt Upgrade Tool 2024.10.25~8c8907c6-r2 (/usr/bin/owut)
ERROR: Sysupgrade type 'combined-efi' should be one of [ "combined", "rootfs" ]
This is a bug in 'owut/2024.10.25~8c8907c6-r2', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function complete_build_info (/usr/bin/owut:1202:84)
  called from function collect_platform (/usr/bin/owut:1314:39)
  called from function collect_all (/usr/bin/owut:1321:19)
  called from anonymous function (/usr/bin/owut:2024:15)

 `           `  ${issue_url}\n`);`
  Near here -------------------^

root@OpenWrt:~# 
```

**Background**
I'm running OpenWrt via `qemu` in a container (https://github.com/AlbrechtL/openwrt-docker). I decided not to use the combined image, instead I use the kernel and the ext4 rootfs as separate images. Maybe it is time to overthink this decision... Anyway, I'm looking how to upgrade an OpenWrt installation without losing the user packages in a `qemu` environment. For that reason I'm following your great `owut` development.

efahl, 2024-11-09:
Hmm, that's an odd one.  This is on the armsr/armv8 target like in the graphic on your GH page?

It looks like an EFI boot system is being detected in the container, but the build is only producing the non-EFI images (see https://downloads.openwrt.org/snapshots/targets/armsr/armv8/, no `...combined-efi...` files).

Could you check inside the container and see if it thinks it's EFI?
```
$ ll /sys/firmware/
drwxr-xr-x    6 root     root             0 Oct 17 11:42 ./
dr-xr-xr-x   13 root     root             0 Oct 17 11:42 ../
drwxr-xr-x    5 root     root             0 Nov  9 14:43 acpi/
drwxr-xr-x    4 root     root             0 Nov  9 14:43 dmi/
drwxr-xr-x    4 root     root             0 Oct 17 11:42 efi/     <<<<<<<<< existence of this dir means "I'm an efi device"
drwxr-xr-x   18 root     root             0 Nov  9 14:43 memmap/
```

As a workaround try this; that final `no` should force it to ignore the above `efi/` check.
```
$ owut check --device armsr/armv8:generic:ext4:no
...
Target         armsr/armv8
Profile        generic
Package-arch   aarch64_generic
Root-FS-type   ext4
Sys-type       combined
...
```

AlbrechtL, 2024-11-10:
> This is on the armsr/armv8 target like in the graphic on your GH page?

Yes. For armsr I'm using the images [generic-kernel.bin](https://downloads.openwrt.org/snapshots/targets/armsr/armv8/openwrt-armsr-armv8-generic-kernel.bin) and https://downloads.openwrt.org/snapshots/targets/armsr/armv8/openwrt-armsr-armv8-generic-ext4-rootfs.img.gz. But I boot OpenWrt via EFI.

```
root@OpenWrt:~# ll /sys/firmware/
drwxr-xr-x    6 root     root             0 Nov 10 13:25 ./
dr-xr-xr-x   12 root     root             0 Nov 10 13:25 ../
drwxr-xr-x    4 root     root             0 Nov 10 13:26 acpi/
drwxr-xr-x    2 root     root             0 Nov 10 13:26 devicetree/
drwxr-xr-x    4 root     root             0 Nov 10 13:26 dmi/
drwxr-xr-x    3 root     root             0 Nov 10 13:25 efi/
-r--------    1 root     root           398 Nov 10 13:26 fdt
```

```
root@OpenWrt:~# owut check --device armsr/armv8:generic:ext4:no --version-to SNAPSHOT
owut - OpenWrt Upgrade Tool 2024.10.25~8c8907c6-r2 (/usr/bin/owut)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         armsr/armv8
Profile        generic
Package-arch   aarch64_generic
Root-FS-type   ext4
Sys-type       combined
Version-from   24.10-SNAPSHOT r28014-c0a50b1d59 (kernel 6.6.58)
Version-to     SNAPSHOT r28025-5f68e24333 (kernel 6.6.59)
Build-FS-type  ext4
Build-at       2024-11-08T09:51:00Z (~52 hours ago)
Image-prefix   openwrt-armsr-armv8-generic
Image-URL      https://downloads.openwrt.org/snapshots/targets/armsr/armv8
Image-file     openwrt-armsr-armv8-generic-ext4-combined.img.gz
Installed      268 packages
Top-level      144 packages
Default         61 packages
User-installed  89 packages (top-level only)

Package version changes:
[...]
```

AlbrechtL, 2024-11-15:
Update: Inside my [OpenWrt Docker image](https://github.com/AlbrechtL/openwrt-docker) I changed to the squashfs combined image (https://downloads.openwrt.org/snapshots/targets/armsr/armv8/openwrt-armsr-armv8-generic-squashfs-combined.img.gz). Unfortunately, `owut` is still not working.

```
root@OpenWrt:~# owut check
owut - OpenWrt Upgrade Tool 2024.10.27~8c7e42f0-r1 (/usr/bin/owut)
ERROR: Sysupgrade type 'combined-efi' should be one of [ "combined", "rootfs" ]
This is a bug in 'owut/2024.10.27~8c7e42f0-r1', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function complete_build_info (/usr/bin/owut:1203:84)
  called from function collect_platform (/usr/bin/owut:1315:39)
  called from function collect_all (/usr/bin/owut:1322:19)
  called from anonymous function (/usr/bin/owut:2025:15)

 `           `  ${issue_url}\n`);`
  Near here -------------------^
```

Maybe `owut` has also in issue with the new apk package manager
```
root@OpenWrt:~# owut check --device armsr/armv8:generic:ext4:no --version-to SNAPSHOT
owut - OpenWrt Upgrade Tool 2024.10.27~8c7e42f0-r1 (/usr/bin/owut)
ERROR: ubus: Not found: Failed to invoke function 'packagelist' on object 'rpc-sys'
 from: 'ubus call rpc-sys packagelist { "all": true }'
This is a bug in 'owut/2024.10.27~8c7e42f0-r1', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function [anonymous function] (/usr/bin/owut:240:58)
  called from function [anonymous function] (/usr/bin/owut:245:74)
  called from function collect_packages (/usr/bin/owut:964:69)
  called from function collect_all (/usr/bin/owut:1339:19)
  called from anonymous function (/usr/bin/owut:2025:15)

 `           `  ${issue_url}\n`);`
  Near here -------------------^
```

```
root@OpenWrt:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r28073-93d005e6bc'
DISTRIB_TARGET='armsr/armv8'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r28073-93d005e6bc'
DISTRIB_TAINTS=''
```

efahl, 2024-11-15:
> ERROR: Sysupgrade type 'combined-efi' should be one of [ "combined", "rootfs" ]

Yeah, on current snapshot builds we're still dealing with that missing rpc call, but this one is something different for which I have not figured out a solution.  

On x86, the existence of that `/sys/firmware/efi/` directory determines whether you should use a `combined` or `combined-efi` image, but it appears not to do so on the `armsr` target.  I need to get clarification from the devs as to why this is the case, and why there is no `combined-efi` image built for that target.

efahl, 2024-11-15:
Now that I'm thinking about it, maybe it's some `qemu` setting???  I'm only familiar enough with it to copy-paste commands until they work, but I was wondering if there's an option that tells you to set up the VM as EFI-boot or bios-boot or something?  

From the missing `combined-efi`, we can assume that OpenWrt thinks the armsr target will only be bios boot, so if there's a way to set "boot-mode" or whatever when creating the VM, that might be the proper solution (where "proper" == "easiest"), emulating the device as OpenWrt currently expects it...

AlbrechtL, 2024-11-15:
> [...] but I was wondering if there's an option that tells you to set up the VM as EFI-boot or bios-boot or something?

A bios mode (compared to x86) is not available on Arm targets. Arm devices are booting via device-tree (usually u-boot as bootloader) or via EFI. armsr stands for [Arm SystemReady](https://www.arm.com/architecture/system-architectures/systemready-certification-program) and is a generic image that boot on all devices that providing EFI-boot. I configured `qemu` to provide an EFI bios to boot the armsr image. So I don't think that this is a `qemu` specific issue. It should appear on all devices that are using this image. But armsr is quite new, for that reason you can't find much system outside. But this will change, for sure!

>  I need to get clarification from the devs as to why this is the case, and why there is no combined-efi image built for that target.

Great idea! The developers who are maintaining the armsr target should know it. What I can say is that the `openwrt-armsr-armv8-generic-squashfs-combined.img` images are only EFI, noting else.

efahl, 2024-11-15:
Ask on IRC, no response, so reported it as https://github.com/openwrt/openwrt/issues/16966

hanvinke, 2024-12-14:
Hi, it seems _owut_ and _luci-app-attendedsysupgrade_ errors have the same root cause: missing SSL functionality. 

Installing just 2 extra libraries seems to make both working without any fetching errors. These are **wget-ssl** and **libpcre2**

[image]

efahl, 2024-12-14:
> Hi, it seems _owut_ and _luci-app-attendedsysupgrade_ errors have the same root cause: missing SSL functionality.

@hanvinke That is a completely different issue than this one.  If you can reproduce it, log a new issue wherever it is appropriate.

efahl, 2025-01-16:
With the merging of https://github.com/openwrt/openwrt/pull/12963, I think this one is done.

@AlbrechtL , could you wait a day or two until things filter through (it's in both SNAPSHOT and 24.10), and give it a test?  Thanks!

AlbrechtL, 2025-01-19:
> With the merging of https://github.com/openwrt/openwrt/pull/12963, I think this one is done.

Oh great! My Actions pipeline https://github.com/AlbrechtL/openwrt-docker/actions is failing for 3 days ;-). Fixed it in https://github.com/AlbrechtL/openwrt-docker/commit/07f4eea91319d13dbb6339760c076a8493238cbf.

> could you wait a day or two until things filter through (it's in both SNAPSHOT and 24.10), and give it a test? Thanks!

Seems to be step further
**x86_64 and 24.10-SNAPSHOT**
```
root@OpenWrt:~# owut check --version-to SNAPSHOT
owut - OpenWrt Upgrade Tool 2025.01.06~e623a900-r1 (/usr/bin/owut)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         x86/64
Profile        generic
Package-arch   x86_64
Root-FS-type   squashfs
Sys-type       combined
Version-from   24.10-SNAPSHOT r28361-3207fe6636 (kernel 6.6.71)
Version-to     SNAPSHOT r28619-7482296ebf (kernel 6.6.71)
Build-FS-type  squashfs
Build-at       2025-01-18T09:36:37Z (~36 hours ago)
Image-prefix   openwrt-x86-64-generic
Image-URL      https://downloads.openwrt.org/snapshots/targets/x86/64
Image-file     openwrt-x86-64-generic-squashfs-combined.img.gz
Installed      247 packages
Top-level      104 packages
Default         43 packages
User-installed  67 packages (top-level only)
```

The image file `openwrt-x86-64-generic-squashfs-combined.img.gz` looks wrong to me but I can't update from 24.10 to master snapshot because of missing packages. For a 24.10 snapshot upgrade test I have to wait a few days because I just repaired the OpenWrt docker container ;-).

**armsr and SNAPSHOT**
```
root@OpenWrt:~# owut check --version-to SNAPSHOT
owut - OpenWrt Upgrade Tool 2025.01.06~e623a900-r1 (/usr/bin/owut)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         armsr/armv8
Profile        generic
Package-arch   aarch64_generic
Root-FS-type   squashfs
Sys-type       combined-efi
Version-from   SNAPSHOT r28632-9454331b7f (kernel 6.6.71)
Version-to     SNAPSHOT r28632-9454331b7f (kernel 6.6.71)
Build-FS-type  squashfs
Build-at       2025-01-19T10:58:06Z (~10 hours ago)
Image-prefix   openwrt-armsr-armv8-generic
Image-URL      https://downloads.openwrt.org/snapshots/targets/armsr/armv8
Image-file     openwrt-armsr-armv8-generic-squashfs-combined-efi.img.gz
Installed      275 packages
Top-level      147 packages
Default         58 packages
User-installed  91 packages (top-level only)
```

Here the image file `openwrt-armsr-armv8-generic-squashfs-combined-efi.img.gz` is correct

AlbrechtL, 2025-01-20:
OK, when booting the `combined-efi.img` on my qemu x86_64 system in bios mode the EFI directory is not existing.

```
root@OpenWrt:~# ll /sys/firmware/
drwxr-xr-x    5 root     root             0 Jan 20 18:38 ./
dr-xr-xr-x   13 root     root             0 Jan 20 18:38 ../
drwxr-xr-x    5 root     root             0 Jan 20 18:39 acpi/
drwxr-xr-x    4 root     root             0 Jan 20 18:39 dmi/
drwxr-xr-x    9 root     root             0 Jan 20 18:39 memmap/
```

Since I control in openwrt-docker the complete virtual machine, I could try to boot OpenWrt in EFI mode instead in bios mode.

efahl, 2025-01-21:
> OK, when booting the `combined-efi.img` on my qemu x86_64 system in bios mode the EFI directory is not existing.

Yup, that's what I'd expect.

All of the recent changes were only for the ARM platform, so nothing has changed for any of the other efi-capable platforms (x86 and RISC-V).  

> update OpenWrt using owut inside qemu

`owut` uses that existence check to determine the image name, so if `/sys/firmware/efi/` exists on-device (i.e., inside the VM), expect the "Image-URL" to be the -efi one.

efahl, 2025-02-06:
Excellent, yes, I think we've got all the various pieces in builds and device settings lined up finally.

Thanks for reporting this and following through with testing!

closed 2025-02-06 by efahl