efahl/owut #22: ERROR: ubus: Not found: Failed to invoke function 'packagelist' on object 'rpc-sys'

aronsky, 2024-11-13:
This issue started after upgrading to the newest snapshot that uses `apk` instead of `opkg`. I managed to screw up the upgrade and ended up with neither `apk` nor `opkg` present, and had to sideload `apk` from a standard snapshot image. Now, managing packages via `apk` works just fine, but `owut` still gives out this error:

```
owut - OpenWrt Upgrade Tool 2024.10.27~8c7e42f0-r1 (/usr/bin/owut)
ERROR: ubus: Not found: Failed to invoke function 'packagelist' on object 'rpc-sys'
 from: 'ubus call rpc-sys packagelist { "all": true }'
This is a bug in 'owut/2024.10.27~8c7e42f0-r1', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function [anonymous function] (/usr/bin/owut:240:58)
  called from function [anonymous function] (/usr/bin/owut:245:74)
  called from function collect_packages (/usr/bin/owut:964:69)
  called from function collect_all (/usr/bin/owut:1339:19)
  called from anonymous function (/usr/bin/owut:2025:15)

 `           `  ${issue_url}\n`);`
  Near here -------------------^
```

Did I miss some configuration that exposes `apk` over `rpc` while sideloading, or is it another issue?

GoetzGoerisch, 2024-11-13:
Seeing the same with `SNAPSHOT (r28070-f84187035e)` and a `owut` package added with `apk`.

Ping @efahl

efahl, 2024-11-13:
> Did I miss some configuration that exposes `apk` over `rpc` while sideloading, or is it another issue?

Nope, you didn't miss anything, the rpc call just hasn't been updated to handle the `apk` database yet.

At this rate, I'm thinking it's maybe a week out?  You may be able to follow progress here https://github.com/openwrt/openwrt/commit/40b8fbaa9754c86480eefc3692c9116a51a64718#commitcomment-149041733  or maybe the forum thread is better  https://forum.openwrt.org/t/201164  In any case, I'll continue blathering on about this stuff until it works.  😁

aronsky, 2024-11-13:
> follow progress here ***@***.***#commitcomment-149041733

Thanks! I just wanted to verify that I didn't miss something. Will wait for
the fix - thanks for the great work you're doing!

trenchcoatjedi, 2024-11-24:
same issue
```
# owut check
owut - OpenWrt Upgrade Tool 2024.11.20~42cd86c2-r1 (/usr/bin/owut)
ERROR: ubus: Not found: Failed to invoke function 'packagelist' on object 'rpc-sys'
 from: 'ubus call rpc-sys packagelist { "all": true }'
This is a bug in 'owut/2024.11.20~42cd86c2-r1', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function [anonymous function] (/usr/bin/owut:240:58)
  called from function [anonymous function] (/usr/bin/owut:245:74)
  called from function collect_packages (/usr/bin/owut:968:67)
  called from function collect_all (/usr/bin/owut:1356:19)
  called from anonymous function (/usr/bin/owut:2042:15)

 `           `  ${issue_url}\n`);`
  Near here -------------------^
# uname -a
Linux 661WRT 6.6.61 #0 SMP Wed Nov 20 08:49:21 2024 armv7l GNU/Linux
# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r28141-52519a59a8'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r28141-52519a59a8'
DISTRIB_TAINTS=''
```

UltraHKR, 2024-11-28:
https://github.com/openwrt/asu/issues/1097

closed 2024-12-02 by efahl