efahl/owut #23: ERROR: Sysupgrade type 'sysupgrade' should be one of [ "nand-factory", "nand-sysupgrade" ]

jperon, 2024-11-14:
On ASRock G10 SNAPSHOT as available [here](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ipq806x%2Fgeneric&id=asrock_g10):

root@OpenWrt:~# owut upgrade
owut - OpenWrt Upgrade Tool 2024.10.27~8c7e42f0-r1 (/usr/bin/owut)
ERROR: Sysupgrade type 'sysupgrade' should be one of [ "nand-factory", "nand-sysupgrade" ]
This is a bug in 'owut/2024.10.27~8c7e42f0-r1', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function complete_build_info (/usr/bin/owut:1203:84)
  called from function collect_platform (/usr/bin/owut:1315:39)
  called from function collect_all (/usr/bin/owut:1322:19)
  called from anonymous function (/usr/bin/owut:2048:15)

 `           `  ${issue_url}\n`);`
  Near here -------------------^

efahl, 2024-11-15:
Just what I need, another bug on top of all the apk drama.  😂

Thanks for finding this, that's a new one.  I had not stumbled onto the `nand-xxx` variants yet.  Should have a fix soon...

closed 2024-11-15 by efahl

efahl, 2024-11-15:
You should be able to try it out, just
```
wget -O /usr/bin/owut https://raw.githubusercontent.com/efahl/owut/main/files/owut
```
and that patches the code in-place (you should see a goofy version number `%%VERSION%%` when you run it).

efahl, 2024-11-15:
Oh, and expect to see another error as `rpc-sys packagelist` is still not working on snapshots...  (https://github.com/efahl/owut/issues/22)

jperon, 2024-11-15:
Thank you, it did exactly what you expected.