efahl/owut #25: Owut error message on a fresh installed router with last SNAPSHOT Release

torrpenn56, 2024-12-05:
Hello everyone
Context mediatek/filogic (banana pi R4) with sysupgrade to the last SNAPSHOT release
apk update and apk add owut:
(1/6) Installing rpcd (2024.12.02~cc9a471c-r1)
Executing rpcd-2024.12.02~cc9a471c-r1.post-install
(2/6) Installing rpcd-mod-rpcsys (2024.12.02~cc9a471c-r1)
Executing rpcd-mod-rpcsys-2024.12.02~cc9a471c-r1.post-install
(3/6) Installing attendedsysupgrade-common (9)
Executing attendedsysupgrade-common-9.post-install
(4/6) Installing rpcd-mod-file (2024.12.02~cc9a471c-r1)
Executing rpcd-mod-file-2024.12.02~cc9a471c-r1.post-install
(5/6) Installing ucode-mod-uclient (2024.10.22~88ae8f20-r1)
Executing ucode-mod-uclient-2024.10.22~88ae8f20-r1.post-install
(6/6) Installing owut (2024.11.29~286d1014-r1)
Executing owut-2024.11.29~286d1014-r1.post-install
OK: 26 MiB in 164 packages

Command failed:

root@OpenWrt:~# owut
Syntax error: Expecting label
In /usr/bin/owut, line 221, byte 2:

 `    default:   [], // Another simple name list.`
      ^-- Near here

efahl, 2024-12-05:
This is very strange, that exact version runs fine on my test machine.  Now when I install a fresh snapshot on a test VM, it fails just as you are showing...

I wonder if `ucode` has changed or something?

efahl, 2024-12-06:
Well, we got a fix in https://github.com/jow-/ucode/commit/9bcd25f54708c2d6bde79f266f1b74d432f4571f, but it's going to take a while to filter through so we can test it...

As  a workaround, are you comfortable with editing the `owut` source on your device?  If so, just comment out the offending line 221, so it looks like this:
```
$ vi /usr/bin/owut

 221         //default:   [], // Another simple name list.

```

robimarko, 2024-12-06:
@efahl Can you please give this a go:
https://github.com/openwrt/openwrt/pull/17191

efahl, 2024-12-06:
@robimarko Is there an easy `git pull something` that applies that to my local repo?

efahl, 2024-12-06:
Nevermind, I just cut'n'pasted the three lines into the Makefile... Easy enough

robimarko, 2024-12-06:
You can just do:
```
wget https://patch-diff.githubusercontent.com/raw/openwrt/openwrt/pull/17191.patch
git am 17191.patch
```

efahl, 2024-12-06:
@torrpenn56 
Fixed by https://github.com/openwrt/openwrt/commit/af6f1a90bbc824bd22814568c5b6990aa082ce8b

As soon as `apk update && apk list ucode` shows `ucode-2024.12.06~209f041f-r1` as available, then `apk upgrade ucode` and the bug will be fixed.

closed 2024-12-06 by efahl