efahl/owut #27: Using OWUT with Resize ROOTFS option results in device configuration loss (default settings).

JustAnotherEndUser, 2025-01-10:
I used OWUT to upgrade a configured x86/64 device from OpenWRT version 24.10.0-rc4 (with default partition size) to 24.10.0-rc5 with the rootfs expansion option.
Specifically, I ran:  'owut upgrade -S 1024'
Although the version upgrade succeeded, upon reboot I noticed that the previous configuration was cleared. ( All settings were back to default, such as LAN IP, blank root password, http was not enabled for LuCI, etc. )
I had made a backup on rc4 immediately before upgrading, so was able to easily restore settings.
The configuration loss was likely a result of the partition resize process.  
I am reporting this in hopes that there is a possible future workaround, or at least as an informational addition to the OWUT documentation / FAQ.  Thank you.

obsy, 2025-01-10:
Did you read https://openwrt.org/docs/guide-user/installation/sysupgrade.owut#expanding_root_file_system ?

efahl, 2025-01-10:
His issue is resolved, but I asked @JustAnotherEndUser to log this to track improvements that at least reduce the need to dig out the wiki article, and best case fix the issues with sysupgrade that cause loss of config when upgrading.

efahl, 2025-01-11:
@dangowrt Quick question: would I be correct in saying that if I detect `/sys/class/block/mtdblock*` on an OpenWrt device, then its root file system cannot be expanded?

I'm on sort of a side quest to this issue; I would like to suppress `--rootfs-size` from appearing in `owut`'s options on devices where it makes no sense, and poking around, that looks like a candidate.  Are there devices with no `mtdblock` but instead have only `ubiblock` devices serving my purpose (i.e., should I look for both)?  Are there others?  (My hardware selection is quite limited, so I'm sort of guessing on all of this.)

Pix13, 2025-02-05:
Replying to myself: indeed it was the issue, i installed directly fro the repo the latest comit ( that includes the revert rootfs stuff) and the upgrade is running

dioni21, 2025-12-30:
Can this be closed?  I'm not sure after reading the previous feature request.

Steltek, 2026-03-20:
Unaware of the issue described here, I just upgraded an x86 VM from 24.10.4 to 25.12.1 while using --rootfs-size 400.
As far as I can see, my config was preserved. (fs is ext4)