efahl/owut #28: Owut does not *see* installed package

marcust, 2025-01-14:
Why is that?

```
root@aquila:~# owut check -a wpad-mesh-mbedtls,iperf3
owut - OpenWrt Upgrade Tool 2025.01.06~e623a900-r1 (/usr/bin/owut)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         mediatek/filogic
Profile        dlink_aquila-pro-ai-m30-a1
Package-arch   aarch64_cortex-a53
Root-FS-type   squashfs
Sys-type       sysupgrade
Version-from   24.10.0-rc5 r28304-6dacba30a7 (kernel 6.6.69)
Version-to     24.10.0-rc5 r28304-6dacba30a7 (kernel 6.6.69)
Build-FS-type  squashfs
Build-at       2025-01-04T21:35:37Z (~228 hours ago)
Image-prefix   openwrt-24.10.0-rc5-mediatek-filogic-dlink_aquila-pro-ai-m30-a1
Image-URL      https://downloads.openwrt.org/releases/24.10.0-rc5/targets/mediatek/filogic
Image-file     openwrt-24.10.0-rc5-mediatek-filogic-dlink_aquila-pro-ai-m30-a1-squashfs-sysupgrade.bin
Installed      164 packages
Top-level       49 packages
Default         36 packages
User-installed  22 packages (top-level only)

Package version changes:
  wpad-mesh-mbedtls              not-installed                              2024.09.15~5ace39b0-r1
1 packages are out-of-date

Default package analysis:
  Default                        Provided-by
  fitblk                         not installed
  nftables                       nftables-json
  wpad-basic-mbedtls             not installed

There are currently package build failures for 24.10.0-rc5 aarch64_cortex-a53:
  Feed: telephony
    freetdm                      Mon Jan 13 09:04:31 2025 - not installed
Failures don't affect this device, details at
  https://downloads.openwrt.org/releases/faillogs-24.10/aarch64_cortex-a53/

It is safe to proceed with an upgrade
root@aquila:~# opkg list-installed | grep wpad-mesh-mbedtls
wpad-mesh-mbedtls - 2024.09.15~5ace39b0-r1

```

I had it upgrade and then my router fell out of the mesh because no mesh support in wpad, if I not explicitly add `wpad-mesh-mbedtls`. Somehow it does not see the installed version?

Also it will just upgrade over and over again with this command line.

efahl, 2025-01-17:
It should see it, just as you suspect.  Could you post up the output from these two commands for me?

```
owut blob
opkg list-installed wpad*
```

Thanks.

marcust, 2025-01-17:
```
root@aquila:~# owut blob
owut {
  "client": "owut/2025.01.06~e623a900-r1",
  "target": "mediatek/filogic",
  "profile": "dlink_aquila-pro-ai-m30-a1",
  "version": "24.10.0-rc5",
  "version_code": "r28304-6dacba30a7",
  "filesystem": "squashfs",
  "diff_packages": true,
  "packages": [
    "iwinfo",
    "liblucihttp",
    "cgi-io",
    "libc",
    "kmod-crypto-gf128",
    "opkg",
    "iw",
    "kmod-crypto-ctr",
    "kmod-nft-fib",
    "vnstat",
    "kmod-nf-flow",
    "kmod-lib-crc-ccitt",
    "getrandom",
    "kmod-phy-aquantia",
    "kmod-pppoe",
    "procd-ujail",
    "base-files",
    "libustream-mbedtls",
    "uboot-envtools",
    "dnsmasq",
    "kmod-crypto-acompress",
    "kmod-mt76-connac",
    "odhcp6c",
    "uci",
    "luci-ssl",
    "dropbear",
    "mtd",
    "odhcpd-ipv6only",
    "kmod-crypto-hw-safexcel",
    "kmod-mt7915e",
    "urandom-seed",
    "ppp",
    "kmod-leds-gpio",
    "kmod-gpio-button-hotplug",
    "logd",
    "luci-app-attendedsysupgrade",
    "ca-certificates",
    "mt7981-wo-firmware",
    "ca-bundle",
    "luci",
    "owut",
    "kmod-mt7981-firmware",
    "kmod-leds-gca230718",
    "kmod-lib-lzo",
    "kernel",
    "iperf3",
    "urngd",
    "ppp-mod-pppoe"
  ]
}

root@aquila:~# opkg list-installed wpad*
wpad-mesh-mbedtls - 2024.09.15~5ace39b0-r1
```

Synchronos, 2025-03-01:
I have the same problem with wpad-mbedtls instead of wpad-basic-mbedtls.
> :\~# owut check -v
> owut - OpenWrt Upgrade Tool 2025.02.18~f9fae6b3-r1 (/usr/bin/owut)
> ASU-Server     https://sysupgrade.openwrt.org
> Upstream       https://downloads.openwrt.org
> Target         mediatek/filogic
> Profile        openwrt_one
> Package-arch   aarch64_cortex-a53
> Root-FS-type   squashfs
> Sys-type       sysupgrade
> Version-from   24.10.0 r28427-6df0e3d02a (kernel 6.6.73)
> Version-to     24.10.0 r28427-6df0e3d02a (kernel 6.6.73)
> Build-FS-type  squashfs
> Build-at       2025-02-03T23:09:37Z (\~621 hours ago)
> Image-prefix   openwrt-24.10.0-mediatek-filogic-openwrt_one
> Image-URL      https://downloads.openwrt.org/releases/24.10.0/targets/mediatek/filogic
> Image-file     openwrt-24.10.0-mediatek-filogic-openwrt_one-squashfs-sysupgrade.itb
> Installed      328 packages
> Top-level       92 packages
> Default         39 packages
> User-installed  63 packages (top-level only)
> 
> Package version changes:
>   banip                               1.5.2-r1                                   1.5.3-r1
>   luci                                25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-attendedsysupgrade         25.057.55135\\~802a31a                       25.059.57589\~8d7a815
>   luci-app-banip                      25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-ddns                       25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-filemanager                25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-firewall                   25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-nft-qos                    25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-nlbwmon                    25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-package-manager            25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-sqm                        25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-uhttpd                     25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-upnp                       25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-app-usteer                     25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-base                           25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-compat                         25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-lib-base                       25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-lib-chartjs                    25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-lib-ip                         25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-lib-jsonc                      25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-lib-nixio                      25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-lib-uqr                        25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-light                          25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-lua-runtime                    25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-mod-admin-full                 25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-mod-network                    25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-mod-status                     25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-mod-system                     25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-proto-ipv6                     25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-proto-ncm                      25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-proto-ppp                      25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-proto-wireguard                25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-ssl                            25.057.55135\~802a31a                       25.059.57589\~8d7a815
>   luci-theme-bootstrap                25.057.55135\~802a31a                       25.059.57589\~8d7a815
> 34 packages are out-of-date
> 
> Default package analysis:
>   Default                             Provided-by
>   dnsmasq                             dnsmasq-full
>   nftables                            nftables-json
>   **wpad-basic-mbedtls                  not installed**
> 
> There are currently package build failures for 24.10.0 aarch64_cortex-a53:
>   Feed: packages
>     netavark                          Sat Mar  1 13:20:26 2025 - not installed
>   Feed: telephony
>     freetdm                           Sat Mar  1 17:50:34 2025 - not installed
[... 87 lines cut ...]
>     "conntrack": "1.4.8-r1",
>     "comgt-ncm": "0.32-r36",
>     "iptraf-ng": "1.2.1-r1",
>     "luci": "25.059.57589\~8d7a815",
>     "owut": "2025.02.18\~f9fae6b3-r1",
>     "kmod-sched-ipset": "6.6.73-r1",
>     "kmod-mt7981-firmware": "6.6.73.2025.01.14\~8e4f72b6-r1",
>     "kmod-ipt-nat6": "6.6.73-r1",
[... cut ...]
>     "luci-lib-uqr": "25.059.57589\~8d7a815",
>     "kmod-phy-airoha-en8811h": "6.6.73-r1",
>     "libfreetype": "2.13.3-r1",
>     "hostapd-common": "2024.09.15\~5ace39b0-r2",
>     "iperf3": "3.17.1-r4",
>     "kmod-ipt-offload": "6.6.73-r1",
>     "urngd": "2023.11.01\~44365eb1-r1",
>     "iftop": "2018.10.03\~77901c8c-r2",
>     "ppp-mod-pppoe": "2.5.1-r1",
>     "dnsmasq-full": "2.90-r4",
>     "luci-proto-ncm": "25.059.57589\~8d7a815"
>   }
> }

> :\~# opkg list-installed wpad-*
> wpad-mbedtls - 2024.09.15~5ace39b0-r2

nlhintz, 2025-04-28:
I ran into a similar problem with "wpad-mesh-mbedtls"; I think I tracked it down to an error in the [rpcd](https://git.openwrt.org/project/rpcd.git) project caused by the Depends or Conflicts lines in the opkg status file being longer than 255 characters.  If you run the command **ubus call rpc-sys packagelist '{ "all": true }'** on your device, the package is missing there as well (you can manually shorten the Depends and Conflicts lines in /usr/lib/opkg/status and then re-run the **ubus call** cmd to show that the long lines are causing problems; make a backup of the opkg status file before you do this though so you can restore it afterwards).

Increasing the read buffer size in sys.c (see attached) would be a quick fix, but perhaps the parsing of the opkg status needs to be re-worked.

[increase-buffer-size.patch.txt](https://github.com/user-attachments/files/19932609/increase-buffer-size.patch.txt)

efahl, 2025-04-28:
> I think I tracked it down to an error in the [rpcd](https://git.openwrt.org/project/rpcd.git) project caused by the Depends or Conflicts lines in the opkg status file being longer than 255 characters.

Well, that's interesting!  It's definitely a bug, you should log an issue over on https://github.com/openwrt/rpcd/issues and I'll pile on, maybe we can get this fixed.  Another issue with rpcd is that it doesn't have an openwrt-24.10 branch, so we need someone to do that so that we can fix both opkg and apk variants.

perrin4869, 2025-07-27:
this issue seems to be solved on version 24.10.2

perrin4869, 2025-07-27:
hm... that's surprising. For more context, I am running `linksys_e8450-ubi`, and `owut blob -v` now lists `wpa-basic-mbedtls` which was missing before, and `owut upgrade` correctly installs it, I just ran an upgrade earlier

nlhintz, 2025-07-27:
I am running Netgear R7800 (ipq806x-generic-netgear_r7800).  After upgrading to 24.10.2, I had to manually re-install "wpad-mesh-mbedtls".  I've attached the output of "opkg list-installed" and "owut blob -v" after the re-install of "wpad-mesh-mbedtls"; note that "wpad-mesh-mbedtls" is missing from the "owut" output.  Manually updating the "/usr/lib/opkg/status" to shorten the "Conflicts" line as described [here](https://github.com/efahl/owut/issues/28#issuecomment-2833743114) fixes the problem.

The problem really depends on whether or not it encounters a line in "/usr/lib/opkg/status" that is exactly 256 characters long.

[installed.txt](https://github.com/user-attachments/files/21455498/installed.txt)

[owut.txt](https://github.com/user-attachments/files/21455500/owut.txt)

closed 2025-09-05 by marcust

efahl, 2025-09-05:
> WARNING: There are 2 missing default packages, confirm this is expected before proceeding

@marcust 

Ooh, re-examining your opening post, I see
```
  fitblk                         not installed
```
That's very bad, you really need `fitblk` on a filogic device or sysupgrade won't work (it just silently does nothing).

But, if the `owut` warning bothers you, and the other "missing defaults" are expected and intentional, you can suppress the check by adding entries to the config.

I do this on my main router, where two default packages don't make any sense, so I just remove them.  Normally, I'd see that same "2 missing..." warning, but they are instead shown as "ignored"
```
$ owut check -v
...
Default package analysis:
  Default                        Provided-by
  dnsmasq                        dnsmasq-full
  kmod-drm-i915                  user ignored
  kmod-dwmac-intel               user ignored
  nftables                       nftables-json
...
```
because of
```
$ cat /etc/config/attendedsysupgrade
...
config owut 'owut'
        ... other stuff ...
        list ignored_defaults 'kmod-drm-i915'
        list ignored_defaults 'kmod-dwmac-intel'
```

And over on my AP, I have `list ignored_defaults 'wpad-basic-mbedtls'` because the packaging mess for the hostapd/wpad stuff doesn't allow for the "provided-by" detection...