efahl/owut #30: owut misinterprets packages prepended with `-`

nariox, 2025-01-24:
I am running `owut upgrade` on 24.10-rc4 and `owut` complains I don't have `uboot-envtools` installed (more specifically, it complains that I don't have `-uboot-envtools` installed), as shown below:
```
Default package analysis:
Default            Provided-by
mode     │  -uboot-envtools  not installed
...
```
The board's definition does have `uboot-envtools` prepended by a dash (see [here](https://github.com/openwrt/openwrt/blob/8e78bc39a38d2a82818bcae467d3bcbbbe8531f3/target/linux/ramips/image/mt7621.mk#L868) for my specific board, but some other boards like `ath79-nec` also have the dash before `uboot-envtools` in their definition), but I am not familiar enough with the tooling to know what that means (build only requirement?). In any case, I believe this is unintentional (or maybe I am misunderstanding something).

efahl, 2025-01-24:
In the context of the makefiles, that dash prefix means "remove it from this target", so your device should _not_ have uboot-envtools installed by default, so if it's there or has a dash somewhere in the package index, that's a problem.  (Just a note: The dash-prefix behavior is also carried on to the image builders, where you'd say `make image PACKAGES="add-this -remove-that"`.)

Is there a package called `mode` installed on your device?  That snippet you quoted looks mangled somehow and I can't quite make out what's going on there.  Could you try running `owut check | cat` and paste the output from that?  (Piping the output through cat suppresses the colorizing and other fancy output, so we can see just the raw text.)

Also, let's see what the package list looks like, do this and paste that whole json blob (should be like 150-200 lines long):
```
$ ubus call rpc-sys packagelist '{"all":true}'
{
        "packages": {
                "terminfo": "6.4-r2",
...
```

nariox, 2025-01-24:
Sorry, I think I messed up when copying the output. Here's a clean output:
```
# owut check | cat
owut - OpenWrt Upgrade Tool 2025.01.06~e623a900-r1 (/usr/bin/owut)
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         ramips/mt7621
Profile        dlink_dir-2640-a1
Package-arch   mipsel_24kc
Root-FS-type   squashfs
Sys-type       sysupgrade
Version-from   24.10.0-rc6 r28388-58d0057481 (kernel 6.6.73)
Version-to     24.10.0-rc6 r28388-58d0057481 (kernel 6.6.73)
Build-FS-type  squashfs
Build-at       2025-01-22T19:52:54Z (~46 hours ago)
Image-prefix   openwrt-24.10.0-rc6-ramips-mt7621-dlink_dir-2640-a1
Image-URL      https://downloads.openwrt.org/releases/24.10.0-rc6/targets/ramips/mt7621
Image-file     openwrt-24.10.0-rc6-ramips-mt7621-dlink_dir-2640-a1-squashfs-sysupgrade.bin
Installed      171 packages
Top-level       45 packages
Default         34 packages
User-installed  20 packages (top-level only)

Package version changes:
All packages are up-to-date

Default package analysis:
  Default                        Provided-by
  -uboot-envtools                not installed
  nftables                       nftables-json
  uboot-envtools                 not installed
  wpad-basic-mbedtls             not installed

There are currently package build failures for 24.10.0-rc6 mipsel_24kc:
  Feed: packages
    netavark                     Wed Jan 22 01:47:59 2025 - not installed
    python-bcrypt                Wed Jan 22 03:17:36 2025 - not installed
    qemu                         Wed Jan 22 02:31:36 2025 - not installed
  Feed: telephony
    freetdm                      Wed Jan 22 04:59:23 2025 - not installed
Failures don't affect this device, details at
  https://downloads.openwrt.org/releases/faillogs-24.10/mipsel_24kc/

There are no changes, upgrade not necessary
```

P.S.: I have `wpad-mbedtls` installed (so `wpad-basic-mbedtls` is not installed, not sure if that line is also expected).

efahl, 2025-01-25:
I've got it all worked out.  The profiles on some devices remove default packages, which I had not seen before, thanks for finding it.  I have a fix that's being tested, should push it up tonight or tomorrow some time.

In any case, it's actually a non-functional bug, only cosmetic, at least for your case, so rest assured that your upgrade has all the packages it needs and should be fine.

closed 2025-01-25 by efahl

nariox, 2025-01-27:
Thank you for working on it.