efahl/owut #31: Use `packages_versions` instead of just `packages` in build request

aparcar, 2025-01-28:
The `packages` argument is only meant for the firmware selector where you request by package name, since `owut` is validating packages, too, it should request the packages with the latest versions.

efahl, 2025-01-28:
I had forgotten about that, thanks for the reminder.

Just did some testing and I can get it working on the `opkg` releases.  Snapshot builds fail because `base-files` isn't in the package index, so we can't get the expected version, due to this: https://github.com/openwrt/openwrt/blob/main/package/Makefile#L134

Seems like with apk there should be a better way to pin a version than hacking up the index...

efahl, 2025-01-28:
> Seems like with apk there should be a better way to pin a version than hacking up the index...

Maybe we should be using the version constraints in `world`?  I think that's their specific purpose?
```
base-files=1652~ea66541032
libc=1.2.5-r4
kernel=6.6.67~4bcc807001e55af2623861553c9748d6-r1
```

aparcar, 2025-01-28:
@dangowrt @ansuel thoughts on this?

Ansuel, 2025-01-28:
Mhhh i need more info on this... yes in theory using package=ver should pin it but this should be done only for essential package... Or I think no package can be updated afterwards

dangowrt, 2025-01-28:
Imho `owut` should always request packages with their version, as those are also shown to the user before the request is carried out. Failing to then actually make sure what was displayed is also what is being built and delivered would be bad UX imho.

efahl, 2025-01-29:
Take note of the three packages, kernel, libc and base-files, that are excluded at index creation time for the `apk` builds:
 https://github.com/openwrt/openwrt/blob/main/package/Makefile#L134

Currently, `owut` sends a build request, gets the manifest back from a build and compares the known-current versions (from the indexes) and those in the manifest, if they don't match, `owut` raises an error.  For the three packages above, it simply skips them, assuming they must be ok.

What Paul was addressing was my use of `packages` in the build request, rather than `packages_versions` (they are mutually exclusive fields).  With the use of `packages`, the build signature on the ASU server doesn't change even when a package is rebuilt, instead the user (occasionally) receives their previously built image from the ASU cache due to this unchanged signature.  `owut` then errors out saying, "pkg error, expected [new version] but got [old version]" (I've gotten a bunch of reports on the forum about this recently with people doing testing and repeating the same build several times).

So, if I use `packages_versions` in the request, the build's signature changes because the package version numbers become part of the hash, and we fire off a new build, `owut` checks the versions between the request and manifest, it all works and users don't complain!

But...  The kernel, libc and base-files have no entries in the indexes, so I can't use `packages_versions` for `apk`-based builds.  (If you don't include `base-files` in the build request, you get an image without the package, which is less than optimal.)

---
So, my suggestion is that _all_ the packages should be listed in the index, and that any with "non-updatable" status (the intent of that makefile stanza) should utilize the standard `apk` constraints mechanism, i.e., version tags in the `world` file.

dangowrt, 2025-01-29:
👍 All packages installing files onto the rootfs (ie. base-files and libc) should be listed in the index. @aparcar @ynezz Why are we currently excluding them?

ynezz, 2025-01-29:
> So, my suggestion is that all the packages should be listed in the index

Yes. Those JSON files should be treated in a generic fashion, not being tied to certain use case/tool.

> and that any with "non-updatable" status (the intent of that makefile stanza) should utilize the standard apk constraints mechanism, i.e., version tags in the world file.

LGTM

aparcar, 2025-01-29:
Okay, let's add all packages to the JSON index and constrain APK via the world file from not upgrading. 

As a side note, kernel and libc where excluded before already and are purely virtual

efahl, 2025-01-29:
> constrain APK via the world file from not upgrading.

If someone is going to look at the generation of the world file, maybe also think about why it contains all packages on the device instead of just the explicitly installed ones  (i.e., there are pure dependents like `libbpf` and `libatomic` in `world` that are never user-installed).  This is causing `ubus call rpc-sys packagelist` to return identical results for both cases of `{"all":true}` or `false` (may affect other things, this is just the one place I've noticed it).

Ref: https://docs.alpinelinux.org/user-handbook/0.1a/Working/apk.html#_world

efahl, 2025-01-29:
Keeping this open until package index changes are done.

I'll add two issues on openwrt:
1) add all packages to index;
2) generate `world` with version constraints.

When those have been closed, I'll close this one.

efahl, 2025-01-29:
https://github.com/openwrt/openwrt/issues/17774
https://github.com/openwrt/openwrt/issues/17775

efahl, 2025-02-02:
Fixed in https://github.com/efahl/owut/commit/bced54ad64fa8cdbc82914532013f97476197d16, pending full solution in https://github.com/openwrt/openwrt/pull/17823

efahl, 2025-02-06:
The snapshots finally picked up https://github.com/openwrt/openwrt/commit/63e178f06740c473e4347dc195f03e7f57ce59a2, so I could test it through the production ASU server.  Combined with https://github.com/efahl/owut/commit/bced54ad64fa8cdbc82914532013f97476197d16 we have the use of `packages_versions` in use by `owut` on all platforms and versions.

closed 2025-02-06 by efahl