efahl/owut #33: Failed to connect to ubus

Gandulf78, 2025-03-15:
Hi there. 
I am trying to check is an update of openwrt is available with the following command:
sudo opkg update  && owut check

Here is the error that I get (randomly):

Downloading https://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt7621/kmods/6.6.73-1-3abe85def815b59c6c75ac1f92135cb6/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/24.10.0/targets/ramips/mt7621/kmods/6.6.73-1-3abe85def815b59c6c75ac1f92135cb6/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
ERROR: ubus: Unable to connect to ubus socket
mod_ubus.connect
This is a bug in 'owut/2025.02.18~f9fae6b3-r1', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function [anonymous function] (/usr/bin/owut:246:58)
  called from function [anonymous function] (/usr/bin/owut:240:43)
  called from anonymous function (/usr/bin/owut:2092:11)

 `           `  ${issue_url}\n`);`
  Near here -------------------^

efahl, 2025-03-15:
`opkg` has nothing to do with `owut`, so you can ignore the first 50 lines or so.

What do you see when you run `ubus` from the command line?  Show me the output of `ubus call system board`.

Gandulf78, 2025-03-16:
Oh I see now. 
su privilege was missing. 
Sorry about that.

efahl, 2025-03-16:
> su privilege was missing.

???  Are you on an OpenWrt router, running as root?

Gandulf78, 2025-03-16:
No. I am using another user.

efahl, 2025-03-17:
`ubus` (and hence `owut`) must be run with root privileges.

closed 2025-03-17 by efahl