efahl/owut #34: owut check reports error on a snapshot build

magandrez, 2025-03-28:
```
$ owut check
ERROR: ubus: Not found: Failed to resolve object name 'rpc-sys'
 from: 'ubus call rpc-sys packagelist { "all": true }'
This is a bug in 'owut/2025.03.14~52e7d44c-r1', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function [anonymous function] (/usr/bin/owut:247:58)
  called from function [anonymous function] (/usr/bin/owut:252:74)
  called from function collect_packages (/usr/bin/owut:1041:67)
  called from function collect_all (/usr/bin/owut:1429:19)
  called from anonymous function (/usr/bin/owut:2145:15)

 `           `  ${issue_url}\n`);`
  Near here -------------------^
```

Info for the build:

Model
GL.iNet GL-AX1800
Platform
qualcommax/ipq60xx
Version
SNAPSHOT (r29096-b242731987)
Date
3/27/2025, 10:44:57 PM

magandrez, 2025-03-28:
At least `dump`, `blob` and `list` commands report the same error.

efahl, 2025-03-28:
What do you see when you run this `ubus` command?
```bash
$ ubus -v list rpc-sys
'rpc-sys' @9501941d
        "packagelist":{"all":"Boolean"}
        "password_set":{"user":"String","password":"String"}
        "upgrade_test":{}
        "upgrade_start":{"keep":"Boolean"}
        "upgrade_clean":{}
        "factory":{}
        "reboot":{}
```

magandrez, 2025-03-28:
> What do you see when you run this `ubus` command?
> 
> $ ubus -v list rpc-sys
> 'rpc-sys' @9501941d
>         "packagelist":{"all":"Boolean"}
>         "password_set":{"user":"String","password":"String"}
>         "upgrade_test":{}
>         "upgrade_start":{"keep":"Boolean"}
>         "upgrade_clean":{}
>         "factory":{}
>         "reboot":{}

A similar result:

$ ubus -v list rpc-sys
'rpc-sys' @43f36a05
	"packagelist":{"all":"Boolean"}
	"password_set":{"user":"String","password":"String"}
	"upgrade_test":{}
	"upgrade_start":{"keep":"Boolean"}
	"upgrade_clean":{}
	"factory":{}
	"reboot":{}

magandrez, 2025-03-28:
This was after reboot, and not having modified WORLD. 🤔

efahl, 2025-03-28:
Very strange.  Did you reboot again after the failure?  The only thing I can think of is that the `rpcd` demon didn't start properly or something like that.  Did you see anything else that didn't work, `apk list` or anything like that?  (I hate when things just magically start working  again without any explanation. 😀)

magandrez, 2025-03-28:
> Very strange.  Did you reboot again after the failure?  The only thing I can think of is that the `rpcd` demon didn't start properly or something like that.  Did you see anything else that didn't [...]

Nope, all good. So strange...I'll keep an eye on the daemon if it misbehaves and report if needed. I think this can be closed. Thanks for the prompt response!

closed 2025-03-29 by efahl