efahl/owut #36: 2025.02.18~f9fae6b3-r1: Can't remove self-contained list of dependencies

opty77, 2025-04-02:
I can't remove self-contained list of dependencies:

```
root@strong:~# echo $owut_remove_list
luci luci-ssl luci-app-package-manager luci-light luci-app-firewall luci-mod-admin-full luci-proto-ipv6 luci-proto-ppp luci-theme-bootstrap uhttpd-mod-ubus luci-mod-network luci-mod-status luci-mod-system uhttpd luci-base cgi-io liblucihttp-ucode rpcd-mod-luci liblucihttp0
root@strong:~# owut check -r "sysklogd $owut_remove_list"
WARNING: Package 'luci-light' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-mod-admin-full' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-proto-ipv6' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-proto-ppp' is a dependency, removal will have no effect on the build
WARNING: Package 'uhttpd-mod-ubus' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-mod-network' is a dependency, removal will have no effect on the build
WARNING: Package 'uhttpd' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-base' is a dependency, removal will have no effect on the build
WARNING: Package 'liblucihttp-ucode' is a dependency, removal will have no effect on the build
WARNING: Package 'rpcd-mod-luci' is a dependency, removal will have no effect on the build
ERROR: Package 'liblucihttp0' is not installed and cannot be removed
ERROR: Errors collecting package data, terminating.
```

(Please ignore sysklogd -- my own package, not publicly available.)

...although opkg would happily remove them:

```
root@strong:~# opkg --noaction remove $owut_remove_list
Removing package luci from root...
Removing package luci-ssl from root...
Removing package luci-app-package-manager from root...
Removing package luci-light from root...
Removing package luci-app-firewall from root...
Removing package luci-mod-admin-full from root...
Removing package luci-proto-ipv6 from root...
Removing package luci-proto-ppp from root...
Removing package luci-theme-bootstrap from root...
Removing package uhttpd-mod-ubus from root...
Removing package luci-mod-network from root...
Removing package luci-mod-status from root...
Removing package luci-mod-system from root...
Removing package uhttpd from root...
Not deleting modified conffile /etc/config/uhttpd.
Removing package luci-base from root...
Not deleting modified conffile /etc/config/luci.
Removing package cgi-io from root...
Removing package liblucihttp-ucode from root...
Removing package rpcd-mod-luci from root...
Removing package liblucihttp0 from root...
```

efahl, 2025-04-02:
Duplicate of #37

closed 2025-04-02 by efahl

opty77, 2025-04-02:
Not sure about duplicate as even with ABI stripped I'm still getting WARNINGs on another device (already run opkg without `--noaction` on the original one):

```
root@tplink:~# echo $owut_remove_list_noabi
libiwinfo-lua liblua liblucihttp libubus-lua luci luci-lib-base luci-ssl liblucihttp-lua lua luci-app-package-manager luci-lib-ip luci-lib-jsonc luci-lib-nixio luci-light luci-app-firewall luci-mod-admin-full luci-proto-ipv6 luci-proto-ppp luci-theme-bootstrap uhttpd-mod-ubus luci-mod-network luci-mod-status luci-mod-system uhttpd luci-base cgi-io liblucihttp-ucode rpcd-mod-luci
root@tplink:~# owut check -r "sysklogd $owut_remove_list_noabi"
WARNING: Package 'liblua' is a dependency, removal will have no effect on the build
WARNING: Package 'liblucihttp-lua' is a dependency, removal will have no effect on the build
WARNING: Package 'lua' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-lib-ip' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-lib-jsonc' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-lib-nixio' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-light' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-mod-admin-full' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-proto-ipv6' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-proto-ppp' is a dependency, removal will have no effect on the build
WARNING: Package 'uhttpd-mod-ubus' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-mod-network' is a dependency, removal will have no effect on the build
WARNING: Package 'uhttpd' is a dependency, removal will have no effect on the build
WARNING: Package 'luci-base' is a dependency, removal will have no effect on the build
WARNING: Package 'liblucihttp-ucode' is a dependency, removal will have no effect on the build
WARNING: Package 'rpcd-mod-luci' is a dependency, removal will have no effect on the build
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         ath79/generic
Profile        tplink_tl-wdr3600-v1
Package-arch   mips_24kc
Version-from   24.10.0 r28427-6df0e3d02a (kernel 6.6.73)
Version-to     24.10.0 r28427-6df0e3d02a (kernel 6.6.73)
4 packages are out-of-date
It is safe to proceed with an upgrade
```

efahl, 2025-04-02:
Ah, ok, so you're concerned about the warnings with this one and the error on the other...

The warnings are basically `owut` saying, "Sorry, I'm not in charge here and here's why..."

Each of those warnings is indicating that there is a dependency that makes removing the package in question a moot action.  This is because `owut` doesn't send a full package list to the ASU server, just the "top-level" ones (those that have no dependents), and you are removing something that's not on that short list.  If something in the top-level list depends on the package you are removing, then the ASU server will put it back into the build; if no top-level dependency exists, it will be left out.  And hence the "no effect" message, since there's no way `owut` can change the ASU/imagebuilder's behavior (plus you wouldn't want to, since it would result in broken package installs).

I figure it's best to warn about this, rather than just silently ignoring it, because someone will do `owut upgrade -r something`, and then see `opkg list something` shows "something" still installed and think it's a bug.

opty77, 2025-04-02:
I believe I was removing top-level ones too because `opkg remove $owut_remove_list` made owut happy:

```
root@strong:~# owut check -r sysklogd
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         ramips/mt7621
Profile        mtc_wr1201
Package-arch   mipsel_24kc
Version-from   24.10.0 r28427-6df0e3d02a (kernel 6.6.73)
Version-to     24.10.0 r28427-6df0e3d02a (kernel 6.6.73)
4 packages are out-of-date
WARNING: There are 2 missing default packages, confirm this is expected before proceeding
It is safe to proceed with an upgrade
```

opty77, 2025-04-03:
BTW, to generate the `$owut_remove_list` I used the following code:

```ash
#!/bin/sh

# opkg-remove-list - list for 'opkg remove'
#
# Created by Opty <opty@zeryk> on 2025-04-01.

if [ ! -x /bin/opkg ]; then
    echo "FATAL: opkg not found or not executable"

    exit 1
fi

# /tmp file template
tft=/tmp/opkg-remove-list

if [ -e $tft-* ]; then
    echo "FATAL: $tft-* exist"

    exit 1
fi

step=0

while IFS= read -r; do
    pkg="$REPLY"
    deps=$(echo `opkg whatdepends $pkg | grep "depends on $pkg$" | awk '{ print $1 }'`)

    if [ "$deps" ]; then
        deps="$deps "
    fi

    echo "$pkg: $deps"
done | tee -a $tft-$step

final_list=""

while [ -s $tft-$step ]; do
    list=$(echo `grep ': $' $tft-$step | cut -d : -f 1`)

    for item in $list; do
        sed_command="$sed_command s/ $item / /;"
    done

    final_list="$final_list $list"
    next_step=$(($step + 1))

    grep -v ': $' $tft-$step | sed "$sed_command" | tee -a $tft-$next_step

    step=$next_step
done

echo "$final_list"
```

opty77, 2025-04-05:
Faster version:

```diff
--- archive/system/2025-04-05/home/opty/bin/001c_opkg-remove-list       2025-04-01 22:21:18.000000000 +0200
+++ bin/opkg-remove-list        2025-04-05 20:19:00.917210079 +0200
@@ -4,8 +4,12 @@
 #
 # Created by Opty <opty@zeryk> on 2025-04-01.
 
-if [ ! -x /bin/opkg ]; then
-    echo "FATAL: opkg not found or not executable"
+opkg_status=/usr/lib/opkg/status
+
+# if [ ! -x /bin/opkg ]; then
+if [ ! -s $opkg_status ]; then
+    # echo "FATAL: opkg not found or not executable"
+    echo "FATAL: $opkg_status not found or not usable"
 
     exit 1
 fi
@@ -23,7 +27,9 @@
 
 while IFS= read -r; do
     pkg="$REPLY"
-    deps=$(echo `opkg whatdepends $pkg | grep "depends on $pkg$" | awk '{ print $1 }'`)
+    # Slow.
+    # deps=$(echo `opkg whatdepends $pkg | grep "depends on $pkg$" | awk '{ print $1 }'`)
+    deps=$(echo `grep -E '^(Package|Depends): ' $opkg_status | grep -EB 1 "^Depends.*(: |, )$pkg(, |$)" | grep '^Package: ' | cut -d \  -f 2`)
     
     if [ "$deps" ]; then
        deps="$deps "
```

@efahl: Despite the WARNINGs owut downloaded *smaller* image.

opty77, 2025-04-06:
Last attempt: What about just `s/will/might/`?

efahl, 2025-04-07:
> Last attempt: What about just `s/will/might/`?

Using one of your examples above:
```
WARNING: Package 'liblua' is a dependency, removal will have no effect on the build
```

Saying "might" would be incorrect, as invoking `owut ... --remove liblua` will have absolutely no effect on the resulting image.  You can tell `owut` to remove it or not, and nothing changes, hence "will" is the proper term here.

opty77, 2025-04-25:
@efahl: I finally got to `owut upgrade` and no opkg-removed packages pulled back, so?