efahl/owut #37: 2025.02.18~f9fae6b3-r1: Some installed packages reported as not

opty77, 2025-04-02:
Some installed packages reported as not:

```
root@strong:~# owut check -r "sysklogd $owut_remove_list"
WARNING: Package 'luci-light' is a dependency, removal will have no effect on the build
[...]
WARNING: Package 'rpcd-mod-luci' is a dependency, removal will have no effect on the build
ERROR: Package 'liblucihttp0' is not installed and cannot be removed
ERROR: Errors collecting package data, terminating.
root@strong:~# opkg info liblucihttp0
Package: liblucihttp0
Version: 2023.03.15~9b5b683f-r1
Depends: libc
Provides: liblucihttp
Status: install ok installed
Section: libs
Architecture: mipsel_24kc
Size: 7433
Filename: liblucihttp0_2023.03.15~9b5b683f-r1_mipsel_24kc.ipk
Description: LuCI HTTP utility library
Installed-Time: 1738624177
```

efahl, 2025-04-02:
`owut` requires the canonical package name, with the ABI version stripped.  I.e., it simply uses the same form as the `rpc-sys` call reports them.
```
$ ubus call rpc-sys packagelist '{"all":true}' | grep libluci
                "liblucihttp": "2023.03.15~9b5b683f-r1",
                "liblucihttp-ucode": "2023.03.15~9b5b683f-r1",
```
So, in your case, the correct package name is `liblucihttp`.

If in doubt, look in `/usr/lib/opkg/status` for the package and see if it has an `ABIversion` property, which will be removed (and is usually also reported as `Provides`):
```
$ grep 'Package: liblucihttp' /usr/lib/opkg/status -A10
Package: liblucihttp0
ABIVersion: 0                 <<<=== here
Version: 2023.03.15~9b5b683f-r1
Depends: libc
Provides: liblucihttp         <<<=== and here
Status: install ok installed
...
```

opty77, 2025-04-02:
Works, thanks!

closed 2025-04-02 by opty77

opty77, 2025-04-03:
> `owut` requires the canonical package name, with the ABI version stripped.

BTW, could you add this to the [OpenWrt Wiki](https://openwrt.org/docs/guide-user/installation/sysupgrade.owut)?