efahl/owut #41: Cannot connect to asu local server

Gandulf78, 2025-04-26:
Hello. 
I've set up an ASU local server. It is server under Synology reverse_proxy. 
I can access it on the openwrt ssh with curl. 
When I try the owut check, I get the following error:

owut check -v
owut - OpenWrt Upgrade Tool 2025.04.08~ef2bfb4d-r1 (/usr/bin/owut)
ERROR: uclient error code=1
  This could be due to the server being down or inaccessible, check
  https://asu.xxx.synology.me/json/v1/overview.json
ERROR: Response status 0 while downloading
  https://asu.xxx.synology.me/json/v1/overview.json

It seems that it is a SSL issue due to libustream-mbedtls20201210 because it is working with libustream-wolfssl20201210.

efahl, 2025-04-28:
Hmm.  If uclient-fetch works, then owut should, too, as they use the same underlying library (libuclient), and use pretty much the same sequence of calls internally.

What does owut do if you use https, but also use the IP address instead of the host name?  (Probably the same, but...)
```
$ uci set attendedsysupgrade.server.url="https://10.0.0.3"
$ owut check
```

Gandulf78, 2025-04-29:
I cannot tell because I am using caddy as web server and Synology as reverse proxy so I don't have actual IP address to put.

erik434, 2025-09-25:
FWIW, I was seeing the same error messages with my local ASU server. I finally tracked down the issue using Wireshark, and it turned out to be failing because my reverse proxy in front of the ASU server only accepted TLS 1.3 connections, but owut was using TLS 1.2, causing the server to reply with "Alert (Level: Fatal, Description: Protocol Version)".

I found a couple workarounds to get owut working in my environment:

- Tweak my reverse proxy config to accept TLS 1.2 in addition to TLS 1.3
- Set attendedsysupgrade.server.url on my router to avoid the reverse proxy, instead going directly to the ASU server (i.e., http://my-asu-server.lan:8000)

Anyway, it seems odd that owut was consistently failing before a workaround, but I could GET the full reverse-proxied URL (https://asu.my.domain/json/v1/overview.json) just fine using curl, wget, or uclient-fetch.

@efahl any ideas about what might be going on here, and how owut might be able to support TLS 1.3? See attached images for the different TLS Client Hello messages as dissected in Wireshark (1st image is owut, 2nd is uclient-fetch).

[image] 

[image]

The critical difference seems to be that uclient-fetch has the supported_versions extension which includes TLS 1.3, so the negotiation can continue, but owut doesn't include this. Does uclient-fetch do some more TLS setup/config that owut doesn't? Or could it be using a different underlying TLS library?

Here are all the TLS/SSL packages I have installed on my router, in case it helps:

opkg list-installed | grep -Pi "ssl|tls"
libmbedtls21 - 3.6.4-r1
libmosquitto-ssl - 2.0.18-r3
libopenssl3 - 3.0.17-r3
libustream-mbedtls20201210 - 2024.07.28~99bd3d2b-r1
luci-ssl - 25.264.71301~3974d8a
px5g-mbedtls - 11
wpad-basic-mbedtls - 2024.09.15~5ace39b0-r2

efahl, 2025-09-25:
> The critical difference seems to be that uclient-fetch has the `supported_versions` extension which includes TLS 1.3, so the negotiation can continue, but owut doesn't include this. Does [...]

Aha, yes, I suspect that `uclient-fetch` has some setup that I'm missing, just need to find and copy it.  Both `owut` and `uclient-fetch` should be using the same underlying TLS as both are using the same `libuclient`, where all that actual work is done.

Thanks a bunch for the analysis, that gives me some good insight as to how to search out the differences.

Might be here?  https://github.com/openwrt/uclient/blob/88ae8f208dd313f69e268234f7db55956aef1cb9/uclient-fetch.c#L845C3-L845C15

erik434, 2025-09-25:
@efahl glorious!! That fixed it. Very interesting that this change affects whether TLS 1.3 is offered or not; seems like a surprising change in behavior, maybe something that should be documented in the underlying library? Though it sounds like maybe there is no documentation!

Either way, seems like a solid fix - thank you for digging deeper to find this fix!

erik434, 2025-09-25:
Oh gosh, I don't really need credit, but github username is fine if you want! I'm just glad I could help 😊

closed 2025-09-27 by efahl