efahl/owut #43: ubus: Not found: Failed to invoke function 'exec' on object 'file'

trebestie, 2025-05-11:
As suggested by the output
I do not know what does it mean
Hope it help anyway

`owut upgrade -S 1024 -r dnsmasq-dhcpv6
ASU-Server     https://sysupgrade.openwrt.org
Upstream       https://downloads.openwrt.org
Target         rockchip/armv8
Profile        friendlyarm_nanopi-r3s
Package-arch   aarch64_generic
Version-from   24.10.0-rc2 r28161-ea17e958b9 (kernel 6.6.63)
Version-to     24.10.1 r28597-0425664679 (kernel 6.6.86)
152 packages are out-of-date
Request hash:
  686842592fe48a4481167b6729482e6050119b253b7d75d7496c68b22392cf93
[...]
Status:   done
Progress: 100s total =   0s in queue + 100s in build
Build succeeded in 100s total =   0s in queue + 100s to build:
Image saved : /tmp/firmware.bin

ERROR: ubus: Not found: Failed to invoke function 'exec' on object 'file'
 from: 'ubus call file exec { "command": "sysupgrade", "params": [ "--test", "/tmp/firmware.bin" ] }'
This is a bug in 'owut/2025.04.08~ef2bfb4d-r1', please report at
  https://github.com/efahl/owut/issues

In [anonymous function](), file /usr/bin/owut, line 103, byte 24:
  called from function [anonymous function] (/usr/bin/owut:247:58)
  called from function [anonymous function] (/usr/bin/owut:252:74)
  called from function [anonymous function] (/usr/bin/owut:261:55)
  called from function sysupgrade (/usr/bin/owut:321:44)
  called from function verify_image (/usr/bin/owut:1880:45)
  called from anonymous function (/usr/bin/owut:2205:20)

 `           `  ${issue_url}\n`);`
  Near here -------------------^
`

My regards

efahl, 2025-05-11:
That sounds like one of the ubus packages isn't installed.  What do you see when you run this opkg command?  Is that `rpcd-mod-file` installed?  (If not, do `opkg update && opkg install rpcd-mod-file`.)

```
$ opkg info rpcd-mod-file
Package: rpcd-mod-file
Version: 2024.09.17~9f4b86e7-r1
Depends: libc, libubus20250102, libubox20240329, rpcd
Status: install ok installed   <<<=== should show 'installed' like it does here
...
```

efahl, 2025-05-11:
Ok, let's try `ubus` directly, what happens if you run this one?
```
$ ubus call file exec '{ "command": "sysupgrade", "params": [ "--test", "/tmp/firmware.bin" ] }'
```

trebestie, 2025-05-11:
`Command failed: Not found`

efahl, 2025-05-11:
Ouch, that looks like `sysupgrade` is missing.  Try running just `sysupgrade` and see if that does anything.

trebestie, 2025-05-11:
Oh oh, It looks like i messed up something big.
Every command returns I/O Error
It's better that I restart with a fresh installation.
Sorry for the noise

closed 2025-05-11 by trebestie

efahl, 2025-05-11:
Yeah, sounds like something bad happened, so it's probably best to restart from scratch.  Good luck!