efahl/owut #44: 'owut list' parsing problem

grrr2, 2025-06-25:
Hi,

I have installed following packages

```
# opkg list-installed | grep nftables
miniupnpd-nftables - 2.3.7-r1
nftables-json - 1.1.1-r1

```
but owut list shows as:
```
# owut list 
-nftables
```

it looks it is a parsing problem.

Owut version is:
```
# owut --help
owut - OpenWrt Upgrade Tool 2025.05.12~8353c4e9-r1 (/usr/bin/owut)
```

Host details:
```
# cat /etc/os-release
NAME="OpenWrt"
VERSION="24.10.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.2"
VERSION_ID="24.10.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28739-d9340319c6"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.2 r28739-d9340319c6"
OPENWRT_BUILD_DATE="1750711236"
```

Thanks.

grrr2, 2025-06-25:
but should not 'miniupnpd-nftables' recognized as valid package?? i miss that one not the pseudo package, i have all the other custom packages listed with owut list, but the miniupnpd-nftables is missing. i thought, maybe incorrectly, that '-nftables' refers to that with a bit of output problem but are you saying it is not related to that at all but it is the result of the pseudo package? then why the miniupnpd-nftables is not shown?

grrr2, 2025-06-25:
please disregard my comment, thanks for the explanation it makes total sense now, i checked again: the "missing" miniupnpd-nftables package is in the list but embedded into the dependant package luci-app-upnp.

efahl, 2025-06-25:
Yeah, you figured it out.  The `owut list` output is just the top-level packages, not the dependencies that get installed by the top-level ones.  Compare the output from these two commands:
```
ubus call rpc-sys packagelist '{"all":false}'
ubus call rpc-sys packagelist '{"all":true}'
```
Former gives me about 90 entries (the top-level packages), latter one shows all packages, so about 300 entries.  The short list is what `owut` uses to tell the server what should be in the build.