efahl/owut #47: packages missing in target version, cannot upgrade

dangowrt, 2025-07-11:
`owut` seems to require that all packages which were previously listed as installed will also be part of the new image. However, only packages in the `/etc/apk/world` file should be considered as mandatory, as all other packages are merely dependencies of those user-selected ones.

It is normal and expected that packages are dropped, change their name, ... and as long as the top-level user selection from `/etc/apk/world` can be installed fine that should not prevent the user from installing an updated image.

dangowrt, 2025-07-11:
In my case here the packages which owut complains about are *not* part of the `/etc/apk/world` set, but only listed as installed in `/lib/apk/db/installed`.

efahl, 2025-07-11:
Ok, turns out that standard builds do the same thing.

@Ansuel , yeah, I found an example.

Do `make menuconfig` and turn on `luci-app-adblock`.  It has a dependency on the `adblock` package, so when you list top-level packages on the device you should see only the luci app and not the dependency.
```bash
builder$ make menuconfig  ->  luci-app-adblock=Y
builder$ make world
...

builder$ grep adblock ./build_dir/target-x86_64_musl/root-x86/etc/apk/world
adblock    <-- hey, you should not be here
luci-app-adblock

builder$ wc ./build_dir/target-x86_64_musl/root-x86/etc/apk/world
 307  307 4020 ./build_dir/target-x86_64_musl/root-x86/etc/apk/world
```
Yow, I only expected like 90 packages to be listed in `world`.

Install the image and now on the device:
```bash
router$ ubus call rpc-sys packagelist '{"all":false}' | grep adblock
                "adblock": "4.4.2-r3",
                "luci-app-adblock": "25.184.61053~5780be0",
```

But then I can fix it on the device by removing and then adding back the top-level package...
```bash
router$ apk del luci-app-adblock adblock
(1/2) Purging luci-app-adblock (25.184.61053~5780be0)
luci-app-adblock-25.184.61053~5780be0.pre-deinstall: Executing script...
(2/2) Purging adblock (4.4.2-r3)
adblock-4.4.2-r3.pre-deinstall: Executing script...
OK: 61 MiB in 291 packages

router$ ubus call rpc-sys packagelist | grep adblock

router$ grep adblock /etc/apk/world

router$ apk add luci-app-adblock
(1/2) Installing adblock (4.4.2-r3)
adblock-4.4.2-r3: installing file to etc/config/adblock.apk-new
adblock-4.4.2-r3.post-install: Executing script...
(2/2) Installing luci-app-adblock (25.192.00988~4715c6a)
luci-app-adblock-25.192.00988~4715c6a.post-install: Executing script...
OK: 62 MiB in 293 packages

router$ grep adblock /etc/apk/world
luci-app-adblock

router$ ubus call rpc-sys packagelist | grep adblock
                "luci-app-adblock": "25.192.00988~4715c6a",
```

dangowrt, 2025-07-11:
It has always been true for images built with the buildroot that **all** packages become user-selected and hence end up in `/etc/apk/world`. This has been true also when we were still using opkg. The reason is simply that when generating an image using the `.config` file the buildroot has no way to know if a package was selected only as a dependency or if the user deliberately selected it.

Please note that the issue I'm reporting here **is unrelated** to that. As I said, the packages owut detects as missing and subsequently refuses to request the image are **not** listed in `/etc/apk/world` and also do **not** show up in `ubus call rpc-sys packagelist '{"all": false}'`.

I have by now worked-around the issue by commenting out the `break` statement in https://github.com/efahl/owut/blob/8353c4e9e2557509be7226c1d06c3b4d679aa3d0/files/owut#L2182 , the result was a successful upgrade with all user-selected packages being present.

efahl, 2025-07-11:
> _not_ part of the `/etc/apk/world` set

Ah, I misinterpreted.  `owut` is saying in the big package analysis list "xxx ... missing to-version" when it existed in the installed build, but no longer exists because of changing names, dependencies, refactoring or whatever.

This is a tough one.  If the "missing to-version" is being reported because of a build problem, then `owut` really doesn't want to submit a build request (that's pretty common, and usually why those messages appear).  If the package is gone for legitimate reasons, then you want to ignore it and carry on.  

But, there's no way for `owut` (or LuCI app or `auc`) to distinguish between those cases unless we add that info to `package_changes` or some similar structure.   (We could do this on-device with `apk info --depends` on the new, yet-to-be-installed packages and chase all that down, but that's not gonna fly.)

This was an issue with the ABI-versioned names before all the recent changes on that front.  You'd have `libustream-mbedtls20241212` installed and it was "deleted" by an ABI version change, so you'd see it as one of those "missing to-version" entries.  The workaround I had people use for that was to simply remove the package in the build request and say "ignore any no-effect messages":

```
$ owut upgrade --remove libustream-mbedtls20241212
...
WARNING: Package 'libustream-mbedtls20241212' is a dependency, removal will have no effect on the build
...
```
Note that the above message can be deduced by `owut` as the package appears in `installed/db`, but not in `world` (via differencing `ubus packagelist` call with all=true and false).

efahl, 2025-07-11:
Sounds good, I'll make these changes: top-level missing == error, dependency missing == warning.  Should be fairly straightforward, I've already got a function `is_top_level(pkg)`...

My original reasoning was that since package build failures are the most frequent cause of those messages, that I'd reduce the load on the ASU server by not submitting build requests that almost always fail.  But, a lot of broken build requests leak through anyhow, as it's often the case that the package indexes are incomplete/corrupted instead of the packages, so people are used to being told "wait a while and try again".

efahl, 2025-07-11:
Aha, good old `jansson`.  That's one of the ABI-version name culprits...

dangowrt, 2025-07-11:
I chose `jansson` just to provide you with an example, it was just the first encounter of `g:openwrt:abiversion` in my `/lib/apk/db/installed` file, and removing that metadata was the easiest way (and also easiest to revert) to have an installed package for which no target version would exist.