efahl/owut #48: Error with proxmox-backup-client in pre-install script

vtcosta, 2025-08-12:
Hi there.. I am trying to integrate my backup solution (Proxmox Backup Server) within my routers upgrade routine.

When I run proxmox-backup-client manually for the first time, it will prompt the following question:
```
fingerprint: xx:xx:xx:xx:xx:xx:xx:xx
Are you sure you want to continue connecting? (y/n): 
```

After replying `y`, it will add the fingerprint to the file `/root/.config/proxmox-backup/fingerprints`.

Subsequent executions of proxmox-backup-client won't ask for the fingerprint confirmation again, because it already exists in that file.

The problem is, when executing proxmox-backup-client inside the pre-install script, I am getting the following error:
```
Starting backup protocol: Tue Aug 12 03:15:21 2025    
certificate validation failed - Certificate fingerprint was not confirmed.
Error: client error (Connect)

Caused by:
    0: error:0A000086:SSL routines:tls_post_process_server_certificate:certificate verify failed:../ssl/statem/statem_clnt.c:2123:
    1: error:0A000086:SSL routines:tls_post_process_server_certificate:certificate verify failed:../ssl/statem/statem_clnt.c:2123:
ERROR: Pre-install script '/etc/owut.d/pre-install.sh' failed with status 1, aborting upgrade
```

I've added a `env` call in the pre-install script, but `USER` is not defined and `HOME` is pointing to /. Bellow is the output:
```
SHLVL=1
HOME=/
TERM=linux
PATH=/usr/sbin:/usr/bin:/sbin:/bin
PWD=/
```

The solution I've got so far is to set the `HOME` environment variable before executing proxmox-backup-client:
```
HOME=/root proxmox-backup-client .....
```

Just wanted to be sure that the environment should really look like that within the pre-install script.

vtcosta, 2025-08-13:
> Another thought, could you pipe "echo y" to whatever is prompting you about the fingerprint? That might let the script succeed without needing to mess with the environment while the system may not be in a fully-set-up state.

That would probably create files in /.cache and /.config. Even tough these files would not be present after the sysupgrade, I would rather stick with a manually confirmed fingerprint.

Guess I'll stick with the `HOME=/root proxmox-backup-client .....` for now. It's been working without issues so far.

I just tough that the HOME should still be /root while running the pre-install script.

Thanks a lot.

efahl, 2025-08-13:
No, you're right, this is an issue with the `ubus` call that I had not considered.  The current code does not pass in the external environment (which is normal and expected, as that is a security hole), but in this context, you'd expect root-like access even in external scripts.

Here's basically what `owut` is doing now:
```
$ ubus call file exec '{"command": "env"}' | jsonfilter -e '$.stdout'
HOME=/
TERM=linux
BOOT_IMAGE=/boot/vmlinuz
PATH=/usr/sbin:/usr/bin:/sbin:/bin
```
But in fact, we should probably be passing the `env` parameter into the `file exec` call, so that the pre-install script behaves as you expect...

Here's a proof-of-concept.  Put on your black hat and look to see if there's anything you think is insecure about running the pre-init script in the user's `env`...

```
$ cat ubus_env.uc
#!/usr/bin/ucode -S

import * as mod_ubus from "ubus";

let command = "env";

// Or do something like this:
// command = "/root/test_script.sh";
// and create a script to run:
/*
echo " #!/bin/sh
env
echo 'doing some stuff...'
" > /root/test_script.sh
chmod +x /root/test_script.sh
*/

let env = getenv();

let ubus = mod_ubus.connect();
let result = ubus.call("file", "exec", {command: command, params:{}, env: env});
ubus.disconnect();

printf("%.2J\n", result);
printf("%s\n", result.stdout);
```

closed 2025-08-16 by efahl